myriadrf/Novena-RF #11: fpga - automatic switching for LMS IOs based on RX/TX state

guruofquality, 2015-02-08:
```VHDL
        --lms control signals
        lms_rxen : out std_logic;
        lms_txen : out std_logic;
        lms_lmsrst : out std_logic;
        mrf_gpio0 : out std_logic;
        mrf_gpio1 : out std_logic;
        mrf_gpio2 : out std_logic;
````

I dont know what the last three gpios are for, perhaps antenna switches. Currently, these IOs are controlled by a SPI-based state machine controlled by the ARM. It may be desirable to control these based on the ADC/DAC active signals coming from the framer/deframer; for the purposes of automatic antenna switching (if that is indeed their use case).

rvadoklis, 2015-02-09:
The RF switches on the RF board (MyraidRF board) are controlled via the GPIO 0-2 logic signals, provided by the FPGA module. This enables the user to choose RF input/output depending
on the operation frequency. The truth table of the GPIO 0-2 settings is shown below.

LMS6002D RF Input/output  |  GPIO 0  |  GPIO 1  |  GPIO 2  |  Description 
--------------------------------------|------------- |-------------|--------------|-----------------
TX out 1  |  X   |  X  |   0   |  High band output (1500 – 3800 MHz)
TX out 2  |  X  |  X  |   1 |   Broadband output
Rx in 1  |  1 |  1  |   X  |  Low band input (300 – 2200 MHz)
Rx in 2  |  0 | 1 |  X |  High band input (1500-3800MHz)
Rx in 3  | 0  |   0  |  X |   Broadband input

Where:
X - don't care,
0 - logic low,
1  - logic high (3.3V for signal level).

Note: LMS6002D RF input/output have to be selected/programmed via SPI registers.

guruofquality, 2015-02-09:
Well if the switches are just for band selection, then there isnt much cause to automatically switch based on transmit state (consider a shared antenna in a half-duplex situation).

But still good to know for the host driver implementation... I will have to see how the lms-suite handles these IOs. Thanks for the information!

guruofquality, 2015-02-14:
Made hooks w/ antenna API to set these switches, we can make it automatic in the future:
https://github.com/myriadrf/Novena-RF/commit/44c31937edb4fa30837316ad0d2debb0711f8df2

closed 2015-02-14 by guruofquality