myriadrf/Novena-RF #22: error while building lms-suite

gasseralaaeldien, 2016-01-31:
Hi All,
while building lms-suite with this command : 
'mkdir build && cd build && cmake .. && cmake --build .'
it outputs : 

-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Found wxWidgets: TRUE  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/gasser.alaaeldien/SVN/zigbee/lms-suite-master/lms-suite-master/LMS6002D/lms-suite/src/build
Scanning dependencies of target lms-suite
[  1%] Building CXX object CMakeFiles/lms-suite.dir/src/CommonUtilities.cpp.o
[  3%] Building CXX object CMakeFiles/lms-suite.dir/src/ConnectionCOM.cpp.o
/home/gasser.alaaeldien/SVN/zigbee/lms-suite-master/lms-suite-master/LMS6002D/lms-suite/src/src/ConnectionCOM.cpp:332:2: warning: #warning Currently searching only for ACM connections [-Wcpp]
 #warning Currently searching only for ACM connections
  ^
[  4%] Building CXX object CMakeFiles/lms-suite.dir/src/ConnectionManager.cpp.o
[  6%] Building CXX object CMakeFiles/lms-suite.dir/src/ConnectionSPI.cpp.o
/home/gasser.alaaeldien/SVN/zigbee/lms-suite-master/lms-suite-master/LMS6002D/lms-suite/src/src/ConnectionSPI.cpp: In function ‘bool IsNovenaBoard()’:
/home/gasser.alaaeldien/SVN/zigbee/lms-suite-master/lms-suite-master/LMS6002D/lms-suite/src/src/ConnectionSPI.cpp:40:30: error: elements of array ‘i2c_msg messages [2]’ have incomplete type
     struct i2c_msg messages[2];
                              ^
/home/gasser.alaaeldien/SVN/zigbee/lms-suite-master/lms-suite-master/LMS6002D/lms-suite/src/src/ConnectionSPI.cpp:40:30: error: storage size of ‘messages’ isn’t known
/home/gasser.alaaeldien/SVN/zigbee/lms-suite-master/lms-suite-master/LMS6002D/lms-suite/src/src/ConnectionSPI.cpp:51:25: error: ‘I2C_M_RD’ was not declared in this scope
     messages[1].flags = I2C_M_RD;
                         ^
make[2]: *** [CMakeFiles/lms-suite.dir/src/ConnectionSPI.cpp.o] Error 1
make[1]: *** [CMakeFiles/lms-suite.dir/all] Error 2
make: *** [all] Error 2

i don't know what is wrong

guruofquality, 2016-01-31:
There is an i2c dev package that seems to interfere. Does this command in the readme fix it?
https://github.com/myriadrf/Novena-RF/blob/master/driver/README.md#dependencies

gasseralaaeldien, 2016-02-01:
i run these two lines : 
sudo apt-get install cmake g++ libusb-1.0-0-dev git
sudo apt-get remove libi2c-dev #fixes build issue

but it still gives me the same error

guruofquality, 2016-02-03:
I know it sounds funny, but if libi2c-dev was not originally installed does ```sudo apt-get install libi2c-dev``` install fix it? (this has been the case on ubuntu for example)